deepdrive/deepdrive #6: On imitation learning code release.

SpencerLo-CMU, 2018-04-09:
Thanks again for this wonderful project!

1. I am curious whether you are planning to release imitation learning code. Right now I can see we are using something like behavior cloning strategy (CNN outputs control signal). Is there any plan to make one step further?

2. I am also curious whether we can get some moving objects (vehicle, pedestrian) in maps, so that we can train the robot to behave safely in such dynamic environment. Besides, is there any plan to build up another larger track/map for learning?

crizCraig, 2018-04-09:
Hi Spencer, thanks, we will be releasing reinforcement learning (assuming that's what you meant??? - behavioral cloning is a type of imitation learning as I understand it) agents in the coming months. You can see our work-in-progress [here](https://github.com/deepdrive/deepdrive/tree/bootstrap) in the bootstrap branch. The idea is to use the current baseline (behavioral cloning based) model to bootstrap the RL model. Specifically we are using PPO from OpenAI's excellent baselines project.

We are also planning to implement moving cars / opposing and same lane traffic in the next month or so. Larger maps would be next, but we will probably hold a competition in the sim before that. Let me know if you have thoughts on that - i.e. would a competition in the current map with traffic added be useful or should we do a larger map first?

crizCraig, 2018-04-11:
Hi Spencer.

1. What types of situations are you looking to generate data for?
2. Thanks. What type of feedback do you mean? Also feel free to start a TODO list and I will merge it in!

closed 2018-06-25 by crizCraig