deepdrive/deepdrive #8: Can't control the vehicle manually

foromer4, 2018-04-23:
Thank you for this cool simulator!
I'm not able to control the vehicle steering in any of the run modes I tried.
I can give it commands through the python client, but when I try to do it directly on the 
running window it just doesn't respond. Some of the command like 'p' for pause do work-
so it's not that the app is stuck or anything like that.

crizCraig, 2018-04-23:
Hi Omer, are you able to press <kbd>m</kbd> to control the car manually? <kdd>p</kpd> may only work in the Unreal Editor, but <kbd>Esc</kbd> should work to pause otherwise.

crizCraig, 2018-04-24:
Pretty sure that <kbd>m</kbd> is what you need as I left it out of the README.md. Sorry about that. Closing for now.

closed 2018-04-24 by crizCraig

foromer4, 2018-04-24:
Thank you - "m" did the trick. would you like me the do a  PR with that fix to the documentation?

crizCraig, 2018-04-24:
Hi Omer, should be fixed already. Thanks though!