deepdrive/deepdrive #9: FR - tracking on other vehicles

foromer4, 2018-04-24:
Is it possible to get id and real world distance to other vehicles in the simulation?

crizCraig, 2018-04-24:
Thanks Omer. It's available in Unreal, but we'd have to add it to the Python API. We're thinking of returning something like this:

```
locomoting_objects:
  types: pedestrians, vehicles, bicycles, motorcycles, animals
  properties: id, type, position, rotation, height, width, length, velocity, rotational velocity, acceleration, up_vector, forward_vector, right_vector, transformation matrix, number_of_passengers, on_road, last_collision_time, position_relative_to_agent
```

There are a couple of ways we could do it. One would be to add an extra API call with a search radius, especially if this call is expensive. The other would be to return it every frame as we do with the existing data. In that case, we'd allow setting the search radius or perhaps only return objects visible to your cameras. I think the first crack at this should just add an API call.

Then for distance to the agent, you could use the length of the `position_relative_to_agent` vector. We may not add all the fields above right away, so are there others, besides distance that you think you will need?

crizCraig, 2018-04-24:
Also, for folks doing computer vision, a property for whether the object was in the camera's view would be useful as well, [perhaps using this](https://answers.unrealengine.com/questions/543515/how-to-detect-if-object-is-in-camera-view.html).

foromer4, 2018-04-25:
Thanks, sounds perfect.

hydrogenpi, 2018-04-27:
The hardware for LZ is 6 x 1080ti. But the weight file is not decided yet. The report (in Chinese) could be found here http://www.yikeweiqi.com/news/topline/44206/