deepdrive/deepdrive #10: Support Python 2

crizCraig, 2018-05-03:
Many folks have large existing codebases in Python 2 so it would be good to not force them to port, especially in order to just use the extension (which is actually in the [deepdrive-sim repo](https://github.com/deepdrive/deepdrive-sim). I've been back and forth on whether to write 2/3 compatible code in this repo, and originally wrote a lot of the code for py2, so it won't be a huge deal. There are many places where I assume floats (I think) and a few `yield from`s in the training code. I've noted the ones caught by Pycharm in my working branch on [this commit](https://github.com/deepdrive/deepdrive/commit/5b40f93ce8b74807d49b1f9a0fbc0d92964183f6) with `TODO(py27)`

Thanks to @travelbureau for bringing to my attention.

cclauss, 2019-11-18:
With 44 days until Python 2 end of life, can we close this issue?  Python 3 has been around for 13+ years.

crizCraig, 2019-11-18:
Yes, good point, we've abandoned this in the code for a while with regards to `future` stuff. Also `deepdrive-api` allows for language agnostic integration.

closed 2019-11-18 by crizCraig