deepdrive/deepdrive #11: Ros Support for Deep Drive

akhil22, 2018-05-05:
Hi, 
I was looking for ROS support for Deep Drive. I understand that the code base is in python and I can use python to write sensor drivers that can publish the messages over ROS but I was wondering whether its a good idea to write camera drivers and Lidar drivers in Python? If its not a good idea then I would like to write some basic sensor drivers in C++ for deep drive.

crizCraig, 2018-05-05:
Hi Akhil. Three ways i can see to do this: 
1. Write them in C++, within our Unreal plugin using our existing source / sink abstraction. That way, if you use your agent in another environment, or a real car that uses ROS, you can simply send the messages there instead of to our sim. Also, for projects that use our plugin in a different simulation, they would get the ability to use ROS.

2. You can also do it in our gym env in Python, which will be easier but slower. The advantage to this would be that the baseline agent and other derivitive agents would work with ROS.

3. The best of both worlds would be to do it in the Python extension (you could use C++ and Python there).

akhil22, 2018-05-06:
Hi crizCrag,
I think I would go with the third approach as I would like to test these algorithms on a Truck we have which uses ROS extensively and It has couple of Lidars which would be slower to simulate in python. Thanks for you detailed response.

crizCraig, 2018-05-06:
Awesome. Let me know if you have any more questions.

closed 2018-05-06 by crizCraig