deepdrive/deepdrive #12: Disable graphics

TharunMohandoss, 2018-05-05:
Hi,
   I am currently running a heavy model using this simulator which requires high dedicated GPU RAM(using my GTX 1060 6GB  RAM). Although alone it runs without issues, it fails when I run it alongside deepdrive due to lack of memory. So, I was wondering if there is any way to just run the simulator without displaying the graphics.
Thanks

sourav-roni, 2018-05-05:
Same query here.

crizCraig, 2018-05-05:
Hi Guys,

**Things that work**
One easy way to have the game consume less memory is to change the resolution and graphics quality in the setting menu to their lowest values. That will not affect the resolution or rendering settings of the cameras sent to your agent, so will not affect the performance of your agents.

You can also mess with Tensorflow's GPU memory options we can be quite heavy handed. We start with 0.8, but you may be able to go lower than that.

One thing[ I'm working on](https://github.com/deepdrive/deepdrive/tree/bootstrap) is [training](https://github.com/deepdrive/tf-models) the baseline with MobileNetV2 rather than AlexNet. This will take 10 to 100x less memory on the agent side of things.

**Things that might work**
Try running the exe with the following environment variables - this will apparently run the game without outputting to the screen. I'm not sure if it will take less memory though.
```SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0```

**Things that didn't work**
I tried changing the `ResX` and `ResY` game command line parameters in Windows, and while this does change the window size, the GPU memory remains unchanged. I also tried setting `-nullRHI` but the shared memory does not initialize (on Windows). Sorry, I just happen to be running Windows this morning, I will check things out in Linux asap.

**Things that should work**
To avoid manually setting this every time, we should persist the settings in the menu across launches. We should also look into having a mode that just doesn't render the main camera, and only optionally renders the agent cameras. Finally, we will also at some point have a way of sending cameras over the network, which will allow you to easily have the game on a separate machine, but this will obviously reduce frame rates.

TharunMohandoss, 2018-05-06:
Thank you for the quick reply.

closed 2018-05-08 by crizCraig

crizCraig, 2018-11-03:
Found a way to completely shut-off rendering for default Unreal camera not sent to agents:

Open the Unreal Console by pressing <kbd>`</kbd> and enter
```
show Rendering
```
This will drastically reduce GPU/CPU processing. To do this automatically, you can do a custom build of deepdrive-sim that executes the console command in a [blueprint or C++](http://api.unrealengine.com/INT/BlueprintAPI/Development/ExecuteConsoleCommand/). Also, if someone wants this integrated into the Python API, let us know.

zlw21gxy, 2019-03-22:
@crizCraig https://hg.libsdl.org/SDL/log?rev=SDL_HINT_CUDA_DEVICE in SDL source code nothing about SDL_HINT_CUDA_DEVICE what version SDL you use?