deepdrive/deepdrive #14: Multiplayer ability

akhil22, 2018-05-11:
Hi,
I was wondering how difficult is to have multiple instances of the simulator running on different machine (something like a multiplayer game). With shared scenes and each vehicle is simulated on a single machine.

crizCraig, 2018-05-11:
Ah, so Unreal handles this quite nicely and has gotten even better recently
due to Fortnite. We would need to use Unreal “replication” on things like
vehicle position and decide the frequency of the syncing. I haven’t looked
much more into than that. This would also be a way to add human driven cars
and see how interactions with autonomous cars play out. A networked
multiplayer sim would also scale really well in terms of environment
complexity in that it wouldn’t be constrained by a single machine, so it’s
a super important effort IMO.

akhil22, 2018-05-11:
Hi Craig,
I was working on this specific problem. I used unity rosbridge and have an implementation that can run simulation on multiple machines sharing the environment. Unity has a very nice high level API which makes it so much easier to write mutiplayer games. I was recently looking at Unreal and It also has a good API to support mutiplyer games. The issue with unity is the fact that you have to use rosbridge to send images which uses websockets and its not fast enough to have 8-10 cameras and also both deepdrive and carla have very good environments so I was thinking of extending the same functionality for deepdrive or carla. Since deepdrive already have a shared memory implementation for transmitting camera images I would start with deepdrive. With Unity I was able to get 10 machines each simulating a single vehicle with ROS interface. The issue here mainly was the delay as the frames are not synchronized you will get 10-20 ms delay in position updates from the clients. I can solve this problem in Unity by running the simulation slower then real time so the delay would not result in too much error in the position as observed by other clients, or having a simulated clock and sharing it across the machines and running the simulation in synchronous manner. Since carla and deep drive both have pretty good environments and a good open source community I thought I should just use Unreal API for networking and replicate the same implementation for deepdrive so I just wanted to ask if anybody is already working on this or have any suggestions or comments about the solution.

crizCraig, 2018-05-11:
Hi @akhil22. So we have a synchronous mode in a [this branch](https://github.com/deepdrive/deepdrive-sim/tree/stepping) of the sim. In Unreal 4.14, the version Deepdrive currently uses, you will see some weirdness in the viewer, but the cameras still render fine. This is fixed in 4.18/4.19 which we will update to next release. 

@dnsmernst will know better - but some [barrier](https://en.wikipedia.org/wiki/Barrier_(computer_science)) on clients in the game [server](https://wiki.unrealengine.com/Networking/Replication#Creating_a_server) would accomplish the centralized stepping you did in Unity. We'd also want a timeout to keep the frame rates sane especially as the number of agents increases. It feels like replication should handle a lot of that for you - perhaps through the help of something like [RepNotify](https://docs.unrealengine.com/en-us/Resources/ContentExamples/Networking/1_4).

crizCraig, 2018-05-11:
Very cool use case btw. Excited to see what you come up with!

akhil22, 2018-05-12:
Hi @crizCraig thanks!! I would look at RepNotify and get started with Deepdrive. I dont think anybody is working on this problem currently.

crizCraig, 2018-05-12:
Sure @akhil22! Yep, no one else is working on multiplayer / multi-agent that I know of.