deepdrive/deepdrive #15: Scene segmentation in deepdrive

sourav-roni, 2018-05-21:
Is there a way to get the scene segmentation directly from the simulator, or do we need to use an external segmentation model to predict the segmentation ?

crizCraig, 2018-05-21:
Hi Sourav, we don't, but it would be fairly straightforward in Unreal with a [material](https://docs.unrealengine.com/en-us/Engine/Rendering/Materials) that colors meshes based on their segmentation class.

sourav-roni, 2018-05-21:
Hi Craig, thanks a lot. I will try out the above approach!!

crizCraig, 2018-05-24:
Sounds good. Let me know if you want me to look into the details some more or if you run into any issues.

raks097, 2020-01-29:
@sourav-roni 
Hi,
Where you able to get it to work?

sourav-roni, 2020-02-18:
@raks097 we used an external scene segmentation method

closed 2020-02-18 by sourav-roni