deepdrive/deepdrive #16: DeepDrive on AWS

adrienrenaud, 2018-06-29:
Is there a documented way to run deepdrive on aws (or other cloud provider) with graphics rendering?
I have installed deepdrive on a deep learning ami (ami-6d720012) and started the simulator with
`$ cd ~/Deepdrive/sim/LinuxNoEditor
 $ source DeepDrive.sh
....
LogInit:Warning: Could not initialize SDL: No available displays
Fatal error: [File:/home/a/src/UnrealEngine/Engine/Source/Runtime/Core/Private/Linux/LinuxApplication.cpp] [Line: 41]
FLinuxApplication::CreateLinuxApplication() : PlatformInitMultimedia() failed, cannot create application instance.
[2018.06.29-09.44.26:886][  0]LogLinux:Error: appError called: Assertion failed: Assertion failed:  [File:/home/a/src/UnrealEngine/Engine/Source/Runtime/Core/Private/Linux/LinuxApplication.cpp] [Line: 41]
FLinuxApplication::CreateLinuxApplication() : PlatformInitMultimedia() failed, cannot create application instance.
`
Any help?

crizCraig, 2018-06-29:
Hi Adrien,

Thanks, I haven't tried things on AWS, but can you please check if the following will work:

```
SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./Deepdrive.sh```

I've also integrated it into a branch [here](https://github.com/deepdrive/deepdrive/tree/headless) which works on my Ubuntu machine.

jbnunn, 2020-02-24:
I got the master branch to run on AWS by using the "NVIDIA Gaming PC - Ubuntu 18.04" AMI and using @crizCraig's `SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./Deepdrive.sh` code above. I'm using NoMachine to connect to the Ubuntu desktop.

closed 2020-02-24 by crizCraig