deepdrive/deepdrive #19: Install problem

winnie2333, 2018-07-26:
When I install the deepdirve by install.py, there is an Error as 

Traceback (most recent call last):
  File "C:/Users/gz04707/deepdrive/config.py", line 69, in <module>
    DEEPDRIVE_DIR = _get_deepdrive_dir()
  File "C:/Users/gz04707/deepdrive/config.py", line 57, in _get_deepdrive_dir
    os.makedirs(ret, exist_ok=True)
  File "E:\anaconda3\envs\tensorflow\lib\os.py", line 220, in makedirs
    mkdir(name, mode)
FileNotFoundError: [WinError 3] 系统找不到指定的路径。: ''

crizCraig, 2018-07-27:
Hmm, have you created the parent directory of the one chosen when answering

> Where would you like to store Deepdrive files i.e. sim binaries (1GB), checkpoints (200MB), recordings, and logs)? [Default - C:/Users/XXX/.deepdrive]

JeffersonH44, 2018-08-14:
I'm getting the same problem, it just don't install the simulator and everything else. 

My output:

```
PS C:\Users\Jefferson\deepdrive> python .\install.py
Checking python version
Checking for valid Tensorflow installation
2018-08-14 10:55:23.510627: I C:\users\nwani\_bazel_nwani\mmtm6wb6\execroot\org_tensorflow\tensorflow\core\platform\cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX AVX2
2018-08-14 10:55:23.714753: I C:\users\nwani\_bazel_nwani\mmtm6wb6\execroot\org_tensorflow\tensorflow\core\common_runtime\gpu\gpu_device.cc:1356] Found device 0 with properties:
name: GeForce GTX 860M major: 5 minor: 0 memoryClockRate(GHz): 1.0195
pciBusID: 0000:01:00.0
totalMemory: 2.00GiB freeMemory: 1.64GiB
2018-08-14 10:55:23.731588: I C:\users\nwani\_bazel_nwani\mmtm6wb6\execroot\org_tensorflow\tensorflow\core\common_runtime\gpu\gpu_device.cc:1435] Adding visible gpu devices: 0
2018-08-14 10:55:24.541010: I C:\users\nwani\_bazel_nwani\mmtm6wb6\execroot\org_tensorflow\tensorflow\core\common_runtime\gpu\gpu_device.cc:923] Device interconnect StreamExecutor with strength 1 edge matrix:
2018-08-14 10:55:24.553697: I C:\users\nwani\_bazel_nwani\mmtm6wb6\execroot\org_tensorflow\tensorflow\core\common_runtime\gpu\gpu_device.cc:929]      0
2018-08-14 10:55:24.565144: I C:\users\nwani\_bazel_nwani\mmtm6wb6\execroot\org_tensorflow\tensorflow\core\common_runtime\gpu\gpu_device.cc:942] 0:   N
2018-08-14 10:55:24.576865: I C:\users\nwani\_bazel_nwani\mmtm6wb6\execroot\org_tensorflow\tensorflow\core\common_runtime\gpu\gpu_device.cc:1053] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 20 MB memory) -> physical GPU (device: 0, name: GeForce GTX 860M, pci bus id: 0000:01:00.0, compute capability: 5.0)
2018-08-14 10:55:24.601622: I C:\users\nwani\_bazel_nwani\mmtm6wb6\execroot\org_tensorflow\tensorflow\core\common_runtime\gpu\gpu_device.cc:1435] Adding visible gpu devices: 0
2018-08-14 10:55:24.612822: I C:\users\nwani\_bazel_nwani\mmtm6wb6\execroot\org_tensorflow\tensorflow\core\common_runtime\gpu\gpu_device.cc:923] Device interconnect StreamExecutor with strength 1 edge matrix:
2018-08-14 10:55:24.624771: I C:\users\nwani\_bazel_nwani\mmtm6wb6\execroot\org_tensorflow\tensorflow\core\common_runtime\gpu\gpu_device.cc:929]      0
2018-08-14 10:55:24.633667: I C:\users\nwani\_bazel_nwani\mmtm6wb6\execroot\org_tensorflow\tensorflow\core\common_runtime\gpu\gpu_device.cc:942] 0:   N
2018-08-14 10:55:24.643468: I C:\users\nwani\_bazel_nwani\mmtm6wb6\execroot\org_tensorflow\tensorflow\core\common_runtime\gpu\gpu_device.cc:1053] Created TensorFlow device (/device:GPU:0 with 20 MB memory) -> physical GPU (device: 0, name: GeForce GTX 860M, pci bus id: 0000:01:00.0, compute capability: 5.0)
Tensorflow is working on the GPU.
Tensorflow 1.8.0 detected - meets min version (1.1)
running command: C:\Users\Jefferson\Anaconda3\python.exe -m pip install -r requirements.txt
Requirement already satisfied: deepdrive in c:\users\jefferson\documents\unrealprojects\deepdrive-sim\plugins\deepdriveplugin\source\deepdrivepython (from -r requirements.txt (line 1)) (2.0.20180805204915)
Requirement already satisfied: scipy in c:\users\jefferson\anaconda3\lib\site-packages (from -r requirements.txt (line 2)) (1.1.0)
Requirement already satisfied: numpy>=1.14.0 in c:\users\jefferson\anaconda3\lib\site-packages (from -r requirements.txt (line 3)) (1.14.5)
Requirement already satisfied: h5py in c:\users\jefferson\anaconda3\lib\site-packages (from -r requirements.txt (line 4)) (2.8.0)
Requirement already satisfied: gym==0.9.6 in c:\users\jefferson\anaconda3\lib\site-packages (from -r requirements.txt (line 5)) (0.9.6)
Requirement already satisfied: arrow in c:\users\jefferson\anaconda3\lib\site-packages (from -r requirements.txt (line 6)) (0.12.1)
Requirement already satisfied: matplotlib in c:\users\jefferson\anaconda3\lib\site-packages (from -r requirements.txt (line 7)) (2.2.2)
Requirement already satisfied: pytest-xdist in c:\users\jefferson\anaconda3\lib\site-packages (from -r requirements.txt (line 8)) (1.22.2)
Requirement already satisfied: requests in c:\users\jefferson\anaconda3\lib\site-packages (from -r requirements.txt (line 9)) (2.19.1)
Requirement already satisfied: clint in c:\users\jefferson\anaconda3\lib\site-packages (from -r requirements.txt (line 10)) (0.5.1)
Requirement already satisfied: pillow in c:\users\jefferson\anaconda3\lib\site-packages (from -r requirements.txt (line 11)) (5.1.0)
Requirement already satisfied: boto in c:\users\jefferson\anaconda3\lib\site-packages (from -r requirements.txt (line 12)) (2.48.0)
Requirement already satisfied: six in c:\users\jefferson\anaconda3\lib\site-packages (from h5py->-r requirements.txt (line 4)) (1.11.0)
Requirement already satisfied: pyglet>=1.2.0 in c:\users\jefferson\anaconda3\lib\site-packages (from gym==0.9.6->-r requirements.txt (line 5)) (1.3.2)
Requirement already satisfied: python-dateutil in c:\users\jefferson\anaconda3\lib\site-packages (from arrow->-r requirements.txt (line 6)) (2.7.3)
Requirement already satisfied: cycler>=0.10 in c:\users\jefferson\anaconda3\lib\site-packages (from matplotlib->-r requirements.txt (line 7)) (0.10.0)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in c:\users\jefferson\anaconda3\lib\site-packages (from matplotlib->-r requirements.txt (line 7)) (2.2.0)
[... cut ...]
Requirement already satisfied: pluggy<0.7,>=0.5 in c:\users\jefferson\anaconda3\lib\site-packages (from pytest>=3.0.0->pytest-xdist->-r requirements.txt (line 8)) (0.6.0)
Requirement already satisfied: apipkg>=1.4 in c:\users\jefferson\anaconda3\lib\site-packages (from execnet>=1.1->pytest-xdist->-r requirements.txt (line 8)) (1.5)
Starting baseline agent
Where would you like to store Deepdrive files (i.e. sim binaries (1GB), checkpoints (200MB), recordings, and logs)? [Default - C:\Users\Jefferson\Deepdrive]
C:\Users\Jefferson\Deepdrive written to C:\Users\Jefferson/.deepdrive\deepdrive_dir
###### ><> ><> PyInit_deepdrive_client <>< <>< ######
```

JeffersonH44, 2018-08-14:
I found it, it seems that this import is now allowing to download the simulator and the weights on deepdrive_gym_env.py

```
import csv
import subprocess
import deepdrive_client # this one
import deepdrive_capture
```

I don't where it takes that import, because there is no source file for that, there's a pip installation that maybe it's missing...

crizCraig, 2018-08-14:
I'm not sure this is the same issue @JeffersonH44 - however `deepdrive_client` is a module installed with the `deepdrive` library. Does `pip install deepdrive` succeed for you?

winnie2333, 2018-09-20:
> interesting, I had to reinstall (I tried to update but it doesn't work) the deepdrive package to make it work... (now is downloading everything). thanks @crizCraig

Oh, I'm so sorry that I noticed your email just now. It's happy to hear that you have solved your problem. And I'm willing to discuss the problem of deepdrive with you when you need it. Progress together :)