deepdrive/deepdrive #20: Problem encountered during the training

Bailey02, 2018-08-28:
Thank you for your great work. 
1. I want to use the data set to train an agent, but the data set you provided are too large. so what's the minimum data set needed to get a satisfying result? 
2. there's some hyper-parameters(like the batch size, the learning rate, the weights of the normalization, the loss function defined, etc.) which could be tuned to make the training faster and better? Since you have got a pretty good trained model, could you give some advice about how to tune the hyper-parameters?

crizCraig, 2018-10-30:
Hi @Bailey02 - so sorry I missed this and am responding late. 

1. You can get 80-90% of the performance with a dataset a tenth the size I imagine. You may need to do some more regularization or shrink the size of the net. The latest version in the `bootstrap` branch trains an MNET2 which I don't even think goes through an entire epoch of training.

2. My method has just been to try lots of tweaks, one at a time, and see how the first 100 or so training steps and handful of evals go. You can get a ton of improvement this way vs the traditional grid or random hyperparameter search which is much more computationally expensive. Granted you'll probably find better hyperparameters with more exhaustive approach. There's something to be said for taking educated guesses and thinking about the interplay between params like batch size and learning rate for example. The MNET2 model trains in an hour or so on my old GTX980 so it's pretty easy to play with things on one machine.

Bailey02, 2018-11-01:
Thank you for explanation. Another issue considering the variables that control the action of the agent during testing. You chose  steering and  throttle. you wrote that network overfits on speed which I encountered too. but you calculated the throttle based on the desired speed and current speed, why not just use the throttle of network outputs?

crizCraig, 2018-11-01:
@Bailey02  - I found the network throttle and speed to not work well. Even when the network does not overfit to throttle and speed, as in the new MNET2 due to batch norm, this is still the case. Adding more training data would help, yes. I think the type of data is the key. I currently collect data by randomly veering the car off center and recording the corrective action. This is necessary to better explore the space and reduce the train/test distribution gap. So doing the same thing for throttle will likely fix the issue. This is the 1st iteration of the DAgger algorithm where the model consists of random weights. Continuing DAgger by using progressively better models to collect data would likely improve steering and throttle even further. For more background I recommend [Berkeley's Deep RL class](http://rail.eecs.berkeley.edu/deeprlcourse/) lecture 2 on imitation learning.