deepdrive/deepdrive #22: Error running install.py

RomainV92, 2018-09-18:
Hi,
I tried installing Deepdrive, following the "README" instructions, but at the end of the installation, just when UE4 starts, it crashes without giving any reason.
I found this in the console, probably the source of the error ;

> Register camera for client 1 227x227 FoV 60 label forward cam 227x227 60 FOV
> RegisterCaptureCameraRequest sent
> Poll timed out
> Waiting for RegisterCaptureCameraResponse, time out
> SharedMemoryImpl_Windows::connect Connected successfully to shared mem Local\DeepDriveCapture with size 157286400
> Successfully connected to Local\DeepDriveCapture with max size of 157286400
> Unknown message with type 0 and Size 0 received
> 2018-09-18 16:31:56,712 - gym_deepdrive.envs.deepdrive_gym_env - ERROR - Failed first step of environment
> 2018-09-18 16:31:56,713 - __main__ - ERROR - Error running agent. Failed first step of environment
> Traceback (most recent call last):
>   File "main.py", line 120, in <module>
>     main()
>   File "main.py", line 90, in main
>     raise e
>   File "main.py", line 72, in main
>     gym_env = deepdrive.start(args.experiment_name, args.env_id, fps=args.fps)
>   File "C:\Users\Innovacion\Documents\deepdrive\deepdrive.py", line 31, in start
>     dd_env.connect(cameras, render)
>   File "C:\Users\Innovacion\Documents\deepdrive\gym_deepdrive\envs\deepdrive_gym_env.py", line 662, in connect
>     self._perform_first_step()
>   File "C:\Users\Innovacion\Documents\deepdrive\gym_deepdrive\envs\deepdrive_gym_env.py", line 672, in _perform_first_step
>     raise RuntimeError(error_msg)
> RuntimeError: Failed first step of environment
> SharedMemoryImpl_Windows::disconnect Disconnecting
> Poll timed out
> Waiting for ReleaseAgentControlResponse, time out
> UnregisterClientRequest sent
> Poll timed out
> Socket closed
> 2018-09-18 16:31:59,042 - gym_deepdrive.envs.deepdrive_gym_env - INFO - Closing sim
> --- Logging error ---
> Traceback (most recent call last):
>   File "C:\Users\Innovacion\Anaconda3\envs\python3\lib\logging\handlers.py", line 71, in emit
>     if self.shouldRollover(record):
>   File "C:\Users\Innovacion\Anaconda3\envs\python3\lib\logging\handlers.py", line 185, in shouldRollover
>     self.stream = self._open()
>   File "C:\Users\Innovacion\Anaconda3\envs\python3\lib\logging\__init__.py", line 1060, in _open
>     return open(self.baseFilename, self.mode, encoding=self.encoding)
> NameError: name 'open' is not defined
> Call stack:
>   File "C:\Users\Innovacion\Documents\deepdrive\gym_deepdrive\envs\deepdrive_gym_env.py", line 512, in __del__
>     self.close()
>   File "C:\Users\Innovacion\Documents\deepdrive\gym_deepdrive\envs\deepdrive_gym_env.py", line 526, in close
>     self.close_sim()
>   File "C:\Users\Innovacion\Documents\deepdrive\gym_deepdrive\envs\deepdrive_gym_env.py", line 208, in close_sim
>     log.info('Closing sim')
> Message: 'Closing sim'
> Arguments: ()

Did someone got this error and found a way to fix it? I don't know what to do, i tried building with UE4.14.3 and UE4.20.2, both leading to the exact same error.
I also tried to build the simulator directly (from deepdrive-sim repo), and it didn't work either (but the error was different). Here is the link to the issue ; https://github.com/deepdrive/deepdrive-sim/issues/14

crizCraig, 2018-09-20:
Can you please paste the logs above this to the start of execution @RomainV92? It looks like the sim is not writing out data to shared memory, but some more log data would be helpful. Also, are you able to start the simulator and see the car driving, etc...?

RomainV92, 2018-09-21:
No, i can't start the simulator nor see the car driving.
Here are the entire logs i get from the console when i run "install.py" (Stored in a file for better readability): 
[LogDeepDrive.txt](https://github.com/deepdrive/deepdrive/files/2404349/LogDeepDrive.txt)

And this is the logfile built during the execution of the installation: 

> 2018-09-21 09:18:05,368 - utils - INFO - Downloading https://github.com/deepdrive/deepdrive-sim.git/sim/deepdrive-sim-windows-2.0.20180302194912.zip to C:\Users\Innovacion\Deepdrive\sim...
> 2018-09-21 09:18:06,067 - __main__ - ERROR - Error running agent. Download URL not accessible https://github.com/deepdrive/deepdrive-sim.git/sim/deepdrive-sim-windows-2.0.20180302194912.zip
> 2018-09-21 09:30:35,336 - utils - INFO - Downloading https://s3-us-west-1.amazonaws.com/deepdrive/sim/deepdrive-sim-windows-2.0.20180302194912.zip to C:\Users\Innovacion\Deepdrive\sim...
> 2018-09-21 09:34:22,394 - utils - INFO - done.
> 2018-09-21 09:34:22,493 - utils - INFO - Unzipping temp file C:\Users\INNOVA~1\AppData\Local\Temp\deepdrive-sim-windows-2.0.20180302194912.zip to C:\Users\Innovacion\Deepdrive\sim...
> 2018-09-21 09:35:08,118 - gym_deepdrive.envs.deepdrive_gym_env - INFO - Starting simulator at C:\Users\Innovacion\Deepdrive\sim\WindowsNoEditor\DeepDrive\Binaries/Win64\DeepDrive-Win64-Shipping.exe (takes a few seconds the first time).
> 2018-09-21 09:35:09,987 - gym_deepdrive.envs.deepdrive_gym_env - WARNING - Connection to environment failed, retry (1/10) in 1 seconds
> 2018-09-21 09:35:12,316 - gym_deepdrive.envs.deepdrive_gym_env - WARNING - Connection to environment failed, retry (2/10) in 3 seconds
> 2018-09-21 09:35:16,034 - gym_deepdrive.envs.deepdrive_gym_env - WARNING - Connection to environment failed, retry (3/10) in 3 seconds
> 2018-09-21 09:35:25,785 - gym_deepdrive.envs.deepdrive_gym_env - ERROR - Failed first step of environment
> 2018-09-21 09:35:25,787 - __main__ - ERROR - Error running agent. Failed first step of environment

dnsmernst, 2018-09-21:
Hi, what happens when you start the DeepDrive manually (and only DeepDrive simulation)? Could you please provide a DeepDrive.log for this attempt?

crizCraig, 2018-09-21:
The DeepDrive.log will be in your sim folder @RomainV92, for example

```
C:\Users\a\DeepDrive\sim\WindowsNoEditor\DeepDrive\Saved\Logs\DeepDrive.log
```
Thanks

RomainV92, 2018-09-24:
Alright, here is what i found:
[DeepDrive.log](https://github.com/deepdrive/deepdrive/files/2409970/DeepDrive.log)

I quickly checked the logs, it seems that the wrong GPU is found (i'm on a laptop and also have an AMD Radeon R5), could it be that the engine can't work with the poor Intel HD graphics family...?

PS: Anyway, i checked AMD control panel to check wich card would be used, and it's locked on "Power saving" mode. I know this is a different problem, but do you know how to change that?
[image]

crizCraig, 2018-09-24:
The sim will definitely not run without GPU acceleration. For your case, it looks like there is a bios setting to force AMD to run all apps with the GPU. https://community.amd.com/thread/219001 - It's also likely helpful to shutdown the Unreal editor as that takes quite a bit of GPU memory. You can check GPU memory in Windows per [this](https://superuser.com/questions/160291/how-can-i-monitor-video-memory-usage/1360976#1360976).

RomainV92, 2018-09-25:
I checked my bios, i have an option to disable the discrete card, and that's it (nothing to disable the UMA, which is the one i want to disable...).
I never launched Unreal Editor before running the installation or the .exe files, i always let your scripts do the job (the only thing that was opened was epic games, but that's no big deal, is it?).
I checked the GPU memory, it just confirmed that he GPU isn't used at all whn i run `DeepDrive.exe`.
So this is a hardware problem here.

crizCraig, 2018-09-25:
So to be clear, this does not look to be a deepdrive issue. However, I'd love to see this work for you, so can you try renaming `deepdrive.exe` to `ue4editor.exe` per this https://www.youtube.com/watch?v=3qRbcR7cYpg - This as we know ue4editor.exe is allowed to run in High performance mode. If that works, we'll need to change some python to launch the correct exe, but that's easy.

RomainV92, 2018-09-26:
Ok it's working now! Thanks a lot for your help!

closed 2018-09-26 by RomainV92