deepdrive/deepdrive #23: Install problem: TypeError: stat: can't specify None for path argument

wxm-lebron, 2018-09-23:
Hi，I am going to install deepdrive in a container created by nvidia-docker, but I got an error. I will be very appreciate if someone can help to fix it.

Here is my system information:
Host operating system: Ubuntu 16.04 LTS
Nvidia-docker 2
Docker version 18.06.1-ce
Image for creating the container: tensorflow/tensorflow:1.7.1-devel-gpu-py3
CUDA/CUDNN version in the tensorflow image: CUDA 9.0 / CUDNN 7.0.5
Python version in the tensorflow image: Python 3.5. 2

Here are my outputs:
python install.py
Checking python version
Checking for valid Tensorflow installation
2018-09-23 01:18:32.203269: I tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-09-23 01:18:32.467970: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1344] Found device 0 with properties:
name: GeForce GTX 1080 Ti major: 6 minor: 1 memoryClockRate(GHz): 1.62
pciBusID: 0000:02:00.0
totalMemory: 10.91GiB freeMemory: 9.13GiB
2018-09-23 01:18:32.735416: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1344] Found device 1 with properties:
name: GeForce GTX 1080 Ti major: 6 minor: 1 memoryClockRate(GHz): 1.62
pciBusID: 0000:81:00.0
totalMemory: 10.91GiB freeMemory: 10.75GiB
2018-09-23 01:18:32.735504: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1423] Adding visible gpu devices: 0, 1
2018-09-23 01:18:33.342924: I tensorflow/core/common_runtime/gpu/gpu_device.cc:911] Device interconnect StreamExecutor with strength 1 edge matrix:
2018-09-23 01:18:33.342996: I tensorflow/core/common_runtime/gpu/gpu_device.cc:917]      0 1
2018-09-23 01:18:33.343006: I tensorflow/core/common_runtime/gpu/gpu_device.cc:930] 0:   N N
2018-09-23 01:18:33.343012: I tensorflow/core/common_runtime/gpu/gpu_device.cc:930] 1:   N N
2018-09-23 01:18:33.343623: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1041] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 111 MB memory) -> physical GPU (device: 0, name: GeForce GTX 1080 Ti, pci bus id: 0000:02:00.0, compute capability: 6.1)
2018-09-23 01:18:33.345522: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1041] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:1 with 111 MB memory) -> physical GPU (device: 1, name: GeForce GTX 1080 Ti, pci bus id: 0000:81:00.0, compute capability: 6.1)
2018-09-23 01:18:33.349324: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1423] Adding visible gpu devices: 0, 1
2018-09-23 01:18:33.349454: I tensorflow/core/common_runtime/gpu/gpu_device.cc:911] Device interconnect StreamExecutor with strength 1 edge matrix:
2018-09-23 01:18:33.349520: I tensorflow/core/common_runtime/gpu/gpu_device.cc:917]      0 1
2018-09-23 01:18:33.349540: I tensorflow/core/common_runtime/gpu/gpu_device.cc:930] 0:   N N
2018-09-23 01:18:33.349570: I tensorflow/core/common_runtime/gpu/gpu_device.cc:930] 1:   N N
2018-09-23 01:18:33.350457: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1041] Created TensorFlow device (/device:GPU:0 with 111 MB memory) -> physical GPU (device: 0, name: GeForce GTX 1080 Ti, pci bus id: 0000:02:00.0, compute capability: 6.1)
2018-09-23 01:18:33.350696: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1041] Created TensorFlow device (/device:GPU:1 with 111 MB memory) -> physical GPU (device: 1, name: GeForce GTX 1080 Ti, pci bus id: 0000:81:00.0, compute capability: 6.1)
Tensorflow is working on the GPU.
Tensorflow 1.7.1 detected - meets min version (1.1)
running command: sudo apt-get install -y python3-tk
Reading package lists...
Building dependency tree...
Reading state information...
python3-tk is already the newest version (3.5.1-1).
0 upgraded, 0 newly installed, 0 to remove and 65 not upgraded.
running command: /usr/bin/python -m pip install -r requirements.txt
Requirement already satisfied: deepdrive in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 1)) (2.0.20180302235456)
Requirement already satisfied: scipy in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 2)) (1.1.0)
Requirement already satisfied: numpy>=1.14.0 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 3)) (1.14.3)
Requirement already satisfied: h5py in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 4)) (2.8.0)
Requirement already satisfied: gym==0.9.6 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 5)) (0.9.6)
Requirement already satisfied: arrow in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 6)) (0.12.1)
Requirement already satisfied: matplotlib in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 7)) (2.2.2)
Requirement already satisfied: pytest-xdist in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 8)) (1.23.0)
Requirement already satisfied: requests in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 9)) (2.19.1)
Requirement already satisfied: clint in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 10)) (0.5.1)
Requirement already satisfied: pillow in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 11)) (5.2.0)
Requirement already satisfied: boto in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 12)) (2.49.0)
Requirement already satisfied: six in /usr/local/lib/python3.5/dist-packages (from h5py->-r requirements.txt (line 4)) (1.11.0)
Requirement already satisfied: pyglet>=1.2.0 in /usr/local/lib/python3.5/dist-packages (from gym==0.9.6->-r requirements.txt (line 5)) (1.3.2)
Requirement already satisfied: python-dateutil in /usr/local/lib/python3.5/dist-packages (from arrow->-r requirements.txt (line 6)) (2.7.2)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in /usr/local/lib/python3.5/dist-packages (from matplotlib->-r requirements.txt (line 7)) (2.2.0)
Requirement already satisfied: pytz in /usr/local/lib/python3.5/dist-packages (from matplotlib->-r requirements.txt (line 7)) (2018.4)
Requirement already satisfied: kiwisolver>=1.0.1 in /usr/local/lib/python3.5/dist-packages (from matplotlib->-r requirements.txt (line 7)) (1.0.1)
Requirement already satisfied: cycler>=0.10 in /usr/local/lib/python3.5/dist-packages (from matplotlib->-r requirements.txt (line 7)) (0.10.0)
Requirement already satisfied: pytest>=3.0.0 in /usr/local/lib/python3.5/dist-packages (from pytest-xdist->-r requirements.txt (line 8)) (3.8.0)
Requirement already satisfied: execnet>=1.1 in /usr/local/lib/python3.5/dist-packages (from pytest-xdist->-r requirements.txt (line 8)) (1.5.0)
Requirement already satisfied: pytest-forked in /usr/local/lib/python3.5/dist-packages (from pytest-xdist->-r requirements.txt (line 8)) (0.2)
Requirement already satisfied: urllib3<1.24,>=1.21.1 in /usr/local/lib/python3.5/dist-packages (from requests->-r requirements.txt (line 9)) (1.23)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python3.5/dist-packages (from requests->-r requirements.txt (line 9)) (2018.8.24)
[... 34 lines cut ...]
    main()
  File "main.py", line 102, in main
    path_follower=args.path_follower, fps=args.fps)
  File "/root/deepdrive/tensorflow_agent/agent.py", line 288, in run
    fps=fps)
  File "/root/deepdrive/deepdrive.py", line 16, in start
    env = gym.make(env)
  File "/usr/local/lib/python3.5/dist-packages/gym/envs/registration.py", line 163, in make
    return registry.make(id)
  File "/usr/local/lib/python3.5/dist-packages/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/usr/local/lib/python3.5/dist-packages/gym/envs/registration.py", line 86, in make
    env = cls(**self._kwargs)
  File "/root/deepdrive/gym_deepdrive/envs/deepdrive_gym_env.py", line 140, in __init__
    utils.ensure_executable(utils.get_sim_bin_path())
  File "/root/deepdrive/utils.py", line 212, in ensure_executable
    st = os.stat(path)
TypeError: stat: can't specify None for path argument
Exception ignored in: <bound method DeepDriveEnv.__del__ of <gym_deepdrive.envs.deepdrive_gym_env.DeepDriveEnv object at 0x7fba05f206d8>>
Traceback (most recent call last):
  File "/root/deepdrive/gym_deepdrive/envs/deepdrive_gym_env.py", line 512, in __del__
    self.close()
  File "/root/deepdrive/gym_deepdrive/envs/deepdrive_gym_env.py", line 521, in close
    deepdrive_client.release_agent_control(self.client_id)
SystemError: <built-in function release_agent_control> returned a result with an error set

crizCraig, 2018-09-23:
Hi @wxm-lebron! Unfortunately, our stable release does not support Docker. Our latest [branch](https://github.com/deepdrive/deepdrive/tree/bootstrap) does support Docker, but is changing rapidly, so is not recommended for the faint of heart. Plan for this branch to hit stable though by the end of the year.

wxm-lebron, 2018-09-23:
Hi @crizCraig! Thanks for your notification. I am looking forward to the Docker-support branch.

closed 2018-09-23 by wxm-lebron