deepdrive/deepdrive #28: Install issue: "Could not find a version that satisfies the requirement deepdrive" on vanilla windows 10

janrovenna, 2019-04-08:
Hi, `python install.py` on a clean windows install (with git, python 3.7, of course), comes up with the following error:

`
RuntimeError: C:\Users\[...]\AppData\Local\Programs\Python\Python37\python.exe -m pip install -r requirements.txt finished with error Could not find a version that satisfies the requirement deepdrive (from -r requirements.txt (line 1)) (from versions: )
No matching distribution found for deepdrive (from -r requirements.txt (line 1))
`

Perhaps related: on a clean ubuntu install, the same error appears and is solved using #18 

Thanks!

crizCraig, 2019-04-09:
Hi @janrovenna - ah so the docs stating that we support Python 3.5+ were written before Python 3.7 was released. If you have 3.5 or 3.6, it will work. If you need 3.7, you can try building [from source](https://github.com/deepdrive/deepdrive-sim#building-the-python-extension) or wait a bit for the fix. I'm working on it here: https://github.com/deepdrive/deepdrive-sim/commit/72ef6b0470a002996faa7ebdbd4068c6824b9b75
Sorry for the confusion!

crizCraig, 2019-04-10:
And in master https://travis-ci.org/deepdrive/deepdrive-sim/jobs/518058105

janrovenna, 2019-04-10:
Hi @crizCraig that's done it, successfully ran deepdrive with python 3.5.4! I'm super impressed how easy deepdrive was to download and install, just this one minor issue! Thanks.

Fixed with https://github.com/deepdrive/deepdrive/commit/d03f64b31551af0fc9d886ead98715ac94368927

closed 2019-04-10 by janrovenna