deepdrive/deepdrive #29: How do you pause (or do fixed-time steps) with the python gym API?

ihexx, 2019-05-20:
Hi, I notice that when I run the example.py script in master, the simulator doesn't run synchronously with python; 
i presume these lines mean it should go forward continuously?
`forward = dd.action(throttle=1, steering=0, brake=0,has_control=True)`
...
`observation, reward, done, info = env.step(forward)`
instead it goes in small bursts once a second or so.

How do I get the simulator to stop and wait for python?

ihexx, 2019-05-20:
Is there some config step I'm missing?

crizCraig, 2019-05-20:
Hi @ihexx ! Unfortunately synchronous stepping is not available in 2.0. Please check the `v3` branch https://github.com/deepdrive/deepdrive/blob/v3/example_sync.py or [this canonical link for posterity](https://github.com/deepdrive/deepdrive/blob/bbc483f6a79db712bff57b386a08b3cf5203e296/example_sync.py) for synchronous stepping.

ihexx, 2019-05-20:
Ah I see, thank you.
I've switched to that branch and redid the install, but I'm having trouble running that script however. Here's a bit of the traceback:

>   File "E:\Projects\BOOM\Simulators\DeepDrive\API\util\ensure_sim.py", line 123, in get_sim_url
>     'in bucket %s' % (bucket_search_str, c.BUCKET_URL))
> RuntimeError: Could not find a sim version matching sim/deepdrive-sim-windows-3.0 in bucket https://s3-us-west-1.amazonaws.com/deepdrive

It was complaining of a duplicate simulator earlier, so I deleted the previous deepdrive simulator files from my home directory, and I used a fresh anaconda environment and redid the install, but it's still stuck on this

crizCraig, 2019-05-20:
Hmm, so we only have the Linux versions up right now. Sorry about that, I will see if we can get you a Windows version, asap.

ihexx, 2019-05-20:
Ah I see, thanks. In the mean time I'll spin up a linux machine and try it out

closed 2019-05-20 by ihexx