deepdrive/deepdrive #32: How can I add vehicles to the map?

pobingwanghai, 2019-07-21:
Thanks for the great work!
I am wondering can i add cars to the env and make them run autonomously?
And how can i change the weather?
I cannot find the document on the web, is there a document to explain how to adopt changes using python scripts?

crizCraig, 2019-07-23:
Hi @pobingwanghai - In v3, you can pass `enable_traffic=True` to `sim.start` as well as randomize the sun speed, month (to change the sun's angle), and shadow level. 

Checkout https://github.com/deepdrive/deepdrive/blob/v3/example_sync.py

Enabling traffic will place a few cars on the track that will overtake if the car in front of them is going too slow and there are no cars coming in the opposite lane.

For more control you can call the methods in world.py directly

https://github.com/deepdrive/deepdrive/blob/d60e4aec3958bddbae61766121dcf77dc952ae08/sim/world.py

For the ultimate control, you can access the entire Unreal API in Python without downloading the editor. 

https://docs.deepdrive.io/v/v3/docs/tutorial/uepy/uepy

pobingwanghai, 2019-07-25:
@crizCraig Thank you, the documents are very helpful.

closed 2019-07-25 by pobingwanghai