deepdrive/deepdrive #33: Seems to have problems connecting to amazon EC2 deepdrive instance

wanfuse123, 2019-07-29:
python3.7/site-packages/gym/envs/registration.py:14: PkgResourcesDeprecationWarning: Parameters to load are deprecated.  Call .resolve and .require separately.
  result = entry_point.load(False)

--------- Simulator not found, downloading ----------
2019-07-29 12:34:33,617 - __main__ - ERROR - Error running agent. list index out of range
Traceback (most recent call last):
  File "main.py", line 120, in <module>
    main()
  File "main.py", line 90, in main
    raise e
  File "main.py", line 72, in main
    gym_env = deepdrive.start(args.experiment_name, args.env_id, fps=args.fps)
  File "/home/steven/car3/deepdrive/deepdrive.py", line 16, in start
    env = gym.make(env)
  File "/home/steven/car3/deepdrive/deepdrive/lib/python3.7/site-packages/gym/envs/registration.py", line 163, in make
    return registry.make(id)
  File "/home/steven/car3/deepdrive/deepdrive/lib/python3.7/site-packages/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/home/steven/car3/deepdrive/deepdrive/lib/python3.7/site-packages/gym/envs/registration.py", line 86, in make
    env = cls(**self._kwargs)
  File "/home/steven/car3/deepdrive/gym_deepdrive/envs/deepdrive_gym_env.py", line 136, in __init__
    url = c.BASE_URL + self.get_latest_sim_file()
  File "/home/steven/car3/deepdrive/gym_deepdrive/envs/deepdrive_gym_env.py", line 252, in get_latest_sim_file
    key=lambda y: y[1])[-1]
IndexError: list index out of range
Exception ignored in: <function DeepDriveEnv.__del__ at 0x7f34ae2487b8>
Traceback (most recent call last):
  File "/home/steven/car3/deepdrive/gym_deepdrive/envs/deepdrive_gym_env.py", line 512, in __del__
    self.close()
  File "/home/steven/car3/deepdrive/gym_deepdrive/envs/deepdrive_gym_env.py", line 521, in close
    deepdrive_client.release_agent_control(self.client_id)
SystemError: <built-in function release_agent_control> returned a result with an error set
(deepdrive) (base) steven@steven-1:~/car3/deepdrive$ git clone https://github.com/deepdrive/deepdrive-sim.git

wanfuse123, 2019-07-29:
(car) (base) steven@steven-1:~/car3/deepdrive$ grep -r "BASE_URL" .
./deepdrive-master/gym_deepdrive/envs/deepdrive_gym_env.py:                    url = c.BASE_URL + self.get_latest_sim_file()
./deepdrive-master/config.py:BASE_URL = 'https://s3-us-west-1.amazonaws.com/deepdrive'
./deepdrive-master/config.py:BASE_WEIGHTS_URL = BASE_URL + '/weights'
Binary file ./gym_deepdrive/envs/__pycache__/deepdrive_gym_env.cpython-37.pyc matches
./gym_deepdrive/envs/deepdrive_gym_env.py:                    url = c.BASE_URL + self.get_latest_sim_file()
Binary file ./__pycache__/config.cpython-37.pyc matches
./config.py:BASE_URL = 'https://s3-us-west-1.amazonaws.com/deepdrive'
./config.py:BASE_WEIGHTS_URL = BASE_URL + '/weights'

crizCraig, 2019-08-02:
Sorry for the late response. This issue is due to missing python 3.7 bindings [here](https://pypi.org/project/deepdrive/2.0.20180302235456/#files) which caused a fallback to an incompatible dev version on your machine, something like `deepdrive==3.0.20190730175903.dev0`, where you want a `2.0` version. So if you run the following things should work.
```
pip uninstall deepdrive
pip install deepdrive==2.0.20190722213951
```

closed 2019-08-02 by crizCraig