deepdrive/deepdrive #35: v3 error, sim crash,

tafseerahmed, 2019-08-14:
Command: python3 main.py --mnet2-baseline --experiment my-baseline-test 

LSB Version:	core-9.20170808ubuntu1-noarch:printing-9.20170808ubuntu1-noarch:security-9.20170808ubuntu1-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.3 LTS
Release:	18.04
Codename:	bionic

 glxinfo | grep OpenGL
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce RTX 2080/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 430.26
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6.0 NVIDIA 430.26
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 430.26
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

```
Address successfully parsed 127.0.0.1:9876
Successfully connected to 127.0.0.1:9876
RegisterClientRequest sent
[2019.08.14-12.09.16:275][129]LogDeepDriveServer: Incoming client connection from 127.0.0.1:33680
RegisterClientResponse received client id 1 max cams 8 capture res 2048 protocol version 3.0.20190528000032
Client id is 1
Register camera for client 1 227x227 FoV 60 label forward cam 227x227 60 FOV
RegisterCaptureCameraRequest sent
[2019.08.14-12.09.16:279][130]LogDeepDriveSimulationServerProxy: Client [1] registered isMaster T
[2019.08.14-12.09.16:279][130]LogDeepDriveClientConnection: [1] Client wants to register reqMaster T isMaster T
[2019.08.14-12.09.16:279][130]LogDeepDriveClientConnection: [1] ServerProtocolVersion:  3.0.20190528000032
[2019.08.14-12.09.16:279][130]LogDeepDriveClientConnection: [1] Client registered.
[2019.08.14-12.09.16:293][130]LogDeepDriveCapture: Register CaptureCameraComponent with id 1
[2019.08.14-12.09.16:293][130]DeepDriveCaptureComponent: UCaptureCameraComponent::InitializeComponent 0x0x7f81e871a4c0 camId 1
[2019.08.14-12.09.16:293][130]LogUMG: Display: Widget Class CaptureCameraRowWidget_C - Loaded Fast Template.
[2019.08.14-12.09.16:293][130]LogDeepDriveServer: Camera registered 1 1
RegisterCaptureCameraResponse received 1 1
Successfully connected to /tmp/deepdrive_shared_memory with max size of 157286400
Created DeepDriveSimulation
ConfigureSimulationRequest sent
[2019.08.14-12.09.16:297][130]LogSlate: Took 0.000073 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoRegular.ufont' (155K)
[2019.08.14-12.09.16:319][132]LogDeepDriveSimulationServer: DeepDriveSimulationServer Incoming connection from 0.0.0.0:0
Assertion failed: LockMode == RLM_WriteOnly [File:/home/ue4/UnrealEngine/Engine/Source/Runtime/VulkanRHI/Private/VulkanTexture.cpp] [Line: 1048] 

Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=68112 LargeMemoryPoolOffset=199248 
Malloc Size=207888 LargeMemoryPoolOffset=407152 
Malloc Size=75780 LargeMemoryPoolOffset=482948 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Connection lost
Socket closed
Waiting for ConfigureSimulationResponse, time out
deepdrive_simulation.time_out: Network time out

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "main.py", line 441, in <module>
    main()
  File "main.py", line 233, in main
    run_tf_based_models(args, camera_rigs, driving_style)
  File "main.py", line 248, in run_tf_based_models
    run_agent(args, camera_rigs)
  File "main.py", line 312, in run_agent
    agent_name=args.agent,)
  File "/home/local/atafseer/Masters/deepdrive/agents/dagger/agent.py", line 362, in run
    agent_name=agent_name)
  File "/home/local/atafseer/Masters/deepdrive/agents/dagger/agent.py", line 484, in setup
    env = start_env()
  File "/home/local/atafseer/Masters/deepdrive/agents/dagger/agent.py", line 482, in start_env
    return sim.start(**(sim_args.to_dict()))
  File "/home/local/atafseer/Masters/deepdrive/sim/__init__.py", line 48, in start
    env = start_local_env(args)
  File "/home/local/atafseer/Masters/deepdrive/sim/__init__.py", line 91, in start_local_env
    connect_to_unreal(_env, args)
  File "/home/local/atafseer/Masters/deepdrive/sim/__init__.py", line 136, in connect_to_unreal
    _env.connect(args.cameras)
  File "/home/local/atafseer/Masters/deepdrive/sim/gym_env.py", line 1154, in connect
    seed=c.rng.randint(1, 10**9)):
SystemError: <built-in function connect> returned a result with an error set

```

The weird thing is Deepdrive.exe when run standalone works fine.

crizCraig, 2019-08-14:
Ah yeah, you are running on Vulkan! Only problem is that we cannot capture as the texture is WriteOnly :) Here's the solution for now https://simdocs.deepdrive.io/v/v3/docs/vulkan 

What version of the sim is in your `~/Deepdrive` folder btw? I thought I disabled Vulkan in the latest release...

tafseerahmed, 2019-08-16:
I didn't change any settings in the config at all, just out of the box ran the test. Yup it was the latest release 😐, btw is it possible to use the weather api from airsim over here.

crizCraig, 2019-08-16:
Ah okay, looks like I hadn't packaged/uploaded! Doh! If you run pull latest and run
```
python main.py --update-sim
```
you'll get a version of the sim that doesn't try to use Vulkan. Thanks for bringing this to my attention @tafseerahmed!

closed 2019-08-16 by crizCraig