deepdrive/deepdrive #36: an eror  occur when i run python install.py

CHonChou, 2019-09-14:
2019-09-14 19:16:13,401 - gym_deepdrive.envs.deepdrive_gym_env - ERROR - Failed first step of environment
2019-09-14 19:16:13,966 - __main__ - ERROR - Error running agent. Failed first step of environment
Traceback (most recent call last):
  File "main.py", line 120, in <module>
    main()
  File "main.py", line 90, in main
    raise e
  File "main.py", line 72, in main
    gym_env = deepdrive.start(args.experiment_name, args.env_id, fps=args.fps)
  File "/home/answer/deepdrive/deepdrive.py", line 31, in start
    dd_env.connect(cameras, render)
  File "/home/answer/deepdrive/gym_deepdrive/envs/deepdrive_gym_env.py", line 662, in connect
    self._perform_first_step()
  File "/home/answer/deepdrive/gym_deepdrive/envs/deepdrive_gym_env.py", line 672, in _perform_first_step
    raise RuntimeError(error_msg)
RuntimeError: Failed first step of environment
Poll timed out
Waiting for ReleaseAgentControlResponse, time out
UnregisterClientRequest sent
Successfully unregistered
Socket closed
2019-09-14 19:16:17,535 - gym_deepdrive.envs.deepdrive_gym_env - INFO - Closing sim
--- Logging error ---
[2019.09.14-11.16.17:571][  2]LogScript:Warning: Accessed None trying to read property VehicleHUD
	AliceGT_C /Game/TuningCars/Maps/Demonstration.Demonstration:PersistentLevel.AliceGT_C_0
	Function /Game/TuningCars/Blueprint/AliceGT/AliceGT.AliceGT_C:UnregisterClient:01A9
[2019.09.14-11.16.17:600][  2]LogScript:Warning: Accessed None
	AliceGT_C /Game/TuningCars/Maps/Demonstration.Demonstration:PersistentLevel.AliceGT_C_0
	Function /Game/TuningCars/Blueprint/AliceGT/AliceGT.AliceGT_C:UnregisterClient:01BE
[2019.09.14-11.16.17:600][  2]LogDeepDriveClientConnection: [1] Client wants to unregister isMaster T
[2019.09.14-11.16.17:602][  2]LogDeepDriveClientConnection: [1] Closed connection successfully T
[2019.09.14-11.16.17:602][  2]LogDeepDriveClientConnection: [1] DeepDriveClientConnection::~DeepDriveClientConnection
Traceback (most recent call last):
  File "/usr/lib/python3.5/logging/handlers.py", line 71, in emit
    if self.shouldRollover(record):
  File "/usr/lib/python3.5/logging/handlers.py", line 185, in shouldRollover
    self.stream = self._open()
  File "/usr/lib/python3.5/logging/__init__.py", line 1037, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
NameError: name 'open' is not defined
Call stack:
  File "/home/answer/deepdrive/gym_deepdrive/envs/deepdrive_gym_env.py", line 512, in __del__
    self.close()
  File "/home/answer/deepdrive/gym_deepdrive/envs/deepdrive_gym_env.py", line 526, in close
    self.close_sim()
  File "/home/answer/deepdrive/gym_deepdrive/envs/deepdrive_gym_env.py", line 208, in close_sim
    log.info('Closing sim')
Message: 'Closing sim'
Arguments: ()

crizCraig, 2019-09-17:
Looking into this now

crizCraig, 2019-09-17:
Can you paste the output of
```
ls -la `cat ~/.deepdrive/deepdrive_dir`
```
please @CHonChou ?