deepdrive/deepdrive #40: Sim error, permissions problem

togaen, 2019-11-16:
The sim won't run for me. When I try to run `python example.py`, I get:

```
13:02 $ python example.py 
Traceback (most recent call last):
  File "example.py", line 1, in <module>
    import sim
  File "/home/maeve/deepdrive/sim/__init__.py", line 11, in <module>
    import logs
  File "/home/maeve/deepdrive/logs.py", line 6, in <module>
    import config as c
  File "/home/maeve/deepdrive/config/__init__.py", line 1, in <module>
    from config.directories import *
  File "/home/maeve/deepdrive/config/directories.py", line 1, in <module>
    from util.get_directories import *
  File "/home/maeve/deepdrive/util/get_directories.py", line 72, in <module>
    _ensure_python_bin_config()
  File "/home/maeve/deepdrive/util/get_directories.py", line 61, in _ensure_python_bin_config
    with open(py_bin, 'w') as _dpbf:
PermissionError: [Errno 13] Permission denied: '/home/maeve/.deepdrive/python_bin'
```
But when I run with `sudo python examle.py` I get:

```
Refusing to run with the root privileges.
Address successfully parsed 127.0.0.1:9876
Couldn't connect to 127.0.0.1:9876
```

crizCraig, 2019-11-16:
Hmmm, did you perhaps run sudo python install? If so, I think this is a
problem with the files created by a privileged process being owned by root.
I’d recommend using miniconda or a virtualenv to avoid needing to install
as root if that is indeed what you did.

togaen, 2019-11-16:
@crizCraig Thanks. I had to install with sudo, the install failed without it.

Are there instructions somewhere for using miniconda or virtualenv? I don't know what those are or how to use them.

crizCraig, 2019-11-16:
Probably a good idea also to delete your ~/Deepdrive/ directory. The one
with the deepdrive-sim-linux... in it as well.

crizCraig, 2019-11-16:
Checkout the cloud instructions for using miniconda

togaen, 2019-11-16:
Does that mean this be running on the cloud, then?

crizCraig, 2019-11-16:
No just this section https://docs.deepdrive.io/docs/tutorial/cloud/cloud#miniconda

togaen, 2019-11-16:
I think there are some steps missing. I ran the miniconda install script, but it just put me back into my shell. What do I do after miniconda is installed?

crizCraig, 2019-11-16:
Next make sure that your ~/Deepdrive folder with `deepdrive-sim-linux...`, `weights`, etc... has your user permissions, or just delete it altogether. Then with your miniconda environment active, rerun the install.

togaen, 2019-11-16:
I still get the permissions error in miniconda when I run `python install.py`:

```
 * Install failed, fix issues below and rerun install.py 

Traceback (most recent call last):
  File "install.py", line 237, in <module>
    raise e
  File "install.py", line 233, in <module>
    main()
  File "install.py", line 123, in main
    import config as c
  File "/home/maeve/deepdrive/config/__init__.py", line 1, in <module>
    from config.directories import *
  File "/home/maeve/deepdrive/config/directories.py", line 1, in <module>
    from util.get_directories import *
  File "/home/maeve/deepdrive/util/get_directories.py", line 72, in <module>
    _ensure_python_bin_config()
  File "/home/maeve/deepdrive/util/get_directories.py", line 61, in _ensure_python_bin_config
    with open(py_bin, 'w') as _dpbf:
PermissionError: [Errno 13] Permission denied: '/home/maeve/.deepdrive/python_bin'
```

crizCraig, 2019-11-16:
Ah yes, we need to delete the `.deepdrive` directory or `chmod` on it as well.

togaen, 2019-11-16:
Now it's working, thanks.

crizCraig, 2019-11-16:
Great, sorry for the rough edges and thanks for the feedback. Updating the docs so folks hopefully avoid future similar issues.

closed 2019-11-16 by crizCraig