deepdrive/deepdrive #42: Cannot run PPO baseline on Kevindale scenarios

subhash, 2019-11-20:
When I run
```
python main.py   --map kevindale_bare  -v --scenario=1 --ppo-baseline
```
the agent defaults to `mnet2 dagger`, but when I include the `agent` arg
```
python main.py   --map kevindale_bare  -v --scenario=1 --ppo-baseline --agent bootstrapped_ppo2
```
the map and scenario change over to the default `canyons`. 

What am I doing wrong here?

subhash, 2019-11-29:
I dug some more and found that the problem is deeper than that. The switch `--ppo-baseline` simply does not work. I had to make two changes to get it to work:

Change default for `--agent` to `None` in `main.py`:
```
    args.add_agent_arg(
        '--agent', nargs='?', default=None,
        help='Agent type (%s, %s, %s)' % (c.DAGGER,
                                          c.DAGGER_MNET2,
                                          c.BOOTSTRAPPED_PPO2))
```

and correct the baseline weights dir in `runtime.py`:
```
PPO_BASELINE_WEIGHTS_DIR = os.path.join(WEIGHTS_DIR,
                                        os.path.join('ppo_baseline_agent_weights', 'checkpoints'))
```

@crizCraig Can you confirm if I am doing the right thing here? If yes, I am happy to send a PR to get this fixed

crizCraig, 2019-11-29:
Hi @subhash - thanks for pointing this out. The following should now work to run the PPO baseline in Kevindale

```
python main.py --ppo-baseline --map=kevindale
```

crizCraig, 2019-11-29:
With regards to resuming training, you can use the `--resume_train` parameter.

closed 2019-11-30 by crizCraig

subhash, 2019-12-02:
Thanks @crizCraig, I managed to run the PPO baseline (had to still include the config param `PPO_RESUME_PATH` because the ppo agent does not yet recognize `resume_train`)

Here's the result on `kevindale_bare, scenario 2`. Is this the expected level of performance the latest baseline is at? I understand that the challenge is in tuning it, but want to make sure I am seeing the correct baseline

[image]

crizCraig, 2019-12-02:
Hi @subhash - yes that looks expected. It's only trained on Canyons which is why it's not staying in the lane :)

subhash, 2019-12-03:
@crizCraig Thanks, makes sense. Otherwise, you wouldn't call it a challenge, would you? :smile:

PR #45 (2019-11-30): Fix ppo baseline
Fixes #42