deepdrive/deepdrive #43: libgcc_s.so.1 must be installed for pthread_cancel to work

Heavy02011, 2019-11-23:
running into this error under ubuntu 18.04 after 
```
python example.py
```
Any ideas how to fix this?

crizCraig, 2019-11-23:
Thanks @Heavy02011 - Haven't seen this before, but perhaps try 
```
sudo apt-get install build-essential
```

Heavy02011, 2019-11-23:
Checked that, it's already installed

Heavy02011, 2019-11-23:
any test routines availabe to check the whole system?

crizCraig, 2019-11-23:
Hmm, `install.py` attempts to do that - but I've not seen this so I think our resident Unreal toolchain expert may have some advice here... @adamrehn any thoughts?

Heavy02011, 2019-11-23:
that's the [install log](https://gist.github.com/Heavy02011/fde87dab95048975558e3e0c6ba7b3f2)

adamrehn, 2019-11-24:
This error is new to me. That file ships with Ubuntu (located at `/lib/x86_64-linux-gnu/libgcc_s.so.1`), so it should always be present, irrespective of the packages that are installed.

Heavy02011, 2019-11-24:
the lib is there
```
(dd) rainer@neuron:/lib/x86_64-linux-gnu$ ll libgcc*
-rw-r--r-- 1 root root 96616 Mai  8  2019 libgcc_s.so.1
```

Heavy02011, 2019-12-08:
any new ideas to solve it?

crizCraig, 2019-12-09:
@Heavy02011 - Did you install under [Miniconda](https://docs.deepdrive.io/docs/miniconda)? If not, you could try that. 

Also, if you wouldn't mind, can you try running 
```
/home/rainer/Deepdrive/deepdrive-sim-linux-3.1.20191114055550/DeepDrive/Binaries/Linux/DeepDrive
```
And see if the sim runs without the Python bindings connecting?

crizCraig, 2019-12-09:
Here are the locations of that file on my machine for example

```c2@rtx1:~$ locate libgcc_s.so.1
/home/c2/.config/google-chrome/Default/Extensions/fahmaaghhglfmonjliepjlchgpgfmobi/1.396.0_0/_platform_specific/x86-64/libgcc_s.so.1
/home/c2/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v12_clang-6.0.1-centos7/x86_64-unknown-linux-gnu/lib64/libgcc_s.so.1
/home/c2/UnrealEngine/Engine/Source/ThirdParty/Breakpad/src/processor/testdata/symbols/libgcc_s.so.1
/home/c2/UnrealEngine/Engine/Source/ThirdParty/Breakpad/src/processor/testdata/symbols/libgcc_s.so.1/18B180F90887D8F8B5C35D185444AF4C0
/home/c2/UnrealEngine/Engine/Source/ThirdParty/Breakpad/src/processor/testdata/symbols/libgcc_s.so.1/18B180F90887D8F8B5C35D185444AF4C0/libgcc_s.so.1.sym
/home/c2/UnrealInstalled/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v12_clang-6.0.1-centos7/x86_64-unknown-linux-gnu/lib64/libgcc_s.so.1
/home/c2/UnrealInstalledOld/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v12_clang-6.0.1-centos7/x86_64-unknown-linux-gnu/lib64/libgcc_s.so.1
/home/c2/anaconda3/envs/bl2/lib/libgcc_s.so.1
/home/c2/anaconda3/envs/botleague/lib/libgcc_s.so.1
/home/c2/anaconda3/envs/ci-hooks/lib/libgcc_s.so.1
/home/c2/anaconda3/envs/dd/lib/libgcc_s.so.1
/home/c2/anaconda3/envs/dd2d/lib/libgcc_s.so.1
/home/c2/anaconda3/envs/deepdrive2/lib/libgcc_s.so.1
/home/c2/anaconda3/envs/gcp-helpers/lib/libgcc_s.so.1
/home/c2/anaconda3/envs/leaderboard-generator/lib/libgcc_s.so.1
/home/c2/anaconda3/envs/proplem-worker/lib/libgcc_s.so.1
/home/c2/anaconda3/envs/py36/lib/libgcc_s.so.1
/home/c2/anaconda3/envs/py37/lib/libgcc_s.so.1
/home/c2/anaconda3/envs/scholarly/lib/libgcc_s.so.1
/home/c2/anaconda3/envs/spud/lib/libgcc_s.so.1
/home/c2/anaconda3/lib/libgcc_s.so.1
/home/c2/anaconda3/pkgs/libgcc-ng-8.2.0-hdf63c60_1/lib/libgcc_s.so.1
/home/c2/anaconda3/pkgs/libgcc-ng-9.1.0-hdf63c60_0/lib/libgcc_s.so.1
/home/c2/snap/chromium/949/.config/chromium/Default/Extensions/fahmaaghhglfmonjliepjlchgpgfmobi/1.396.0_0/_platform_specific/x86-64/libgcc_s.so.1
/home/c2/snap/chromium/958/.config/chromium/Default/Extensions/fahmaaghhglfmonjliepjlchgpgfmobi/1.396.0_0/_platform_specific/x86-64/libgcc_s.so.1
/lib/i386-linux-gnu/libgcc_s.so.1
/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/core/8039/lib/i386-linux-gnu/libgcc_s.so.1
/snap/core/8039/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/core/8213/lib/i386-linux-gnu/libgcc_s.so.1
/snap/core/8213/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/core18/1265/lib/i386-linux-gnu/libgcc_s.so.1
/snap/core18/1265/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/core18/1279/lib/i386-linux-gnu/libgcc_s.so.1
/snap/core18/1279/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/vlc/1049/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/vlc/770/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/wine-platform-runtime/51/lib/i386-linux-gnu/libgcc_s.so.1
/snap/wine-platform-runtime/54/lib/i386-linux-gnu/libgcc_s.so.1
/usr/lib/gcc/x86_64-linux-gnu/7/libgcc_s.so.1
```

So it's in the env as well. You could likely add it to LD_LIBRARY_PATH and get past this particular error, but I think Miniconda is going to be best bet to get things working for you.

Heavy02011, 2020-02-03:
thx Craig, will test that!

vasica38, 2020-02-06:
@Heavy02011 did you work out ?

Heavy02011, 2020-12-31:
Sorry for the long response, tried again still the same when doing 
```
(dd) rainer@neuron:~/Deepdrive/deepdrive-sim-linux-3.1.20200229010300/DeepDrive/Binaries/Linux$ ./DeepDrive
```
I get
```
[2020.12.31-18.22.22:955][  0]LogDeepDriveConnectionListener: DeepDriveConnectionListener::Run Socket is listening T
[2020.12.31-18.22.22:955][  0]LogUMG: Display: Widget Class LoadingScreenWidget_C - Loaded Fast Template.
[2020.12.31-18.22.22:961][  0]LogLoad: Took 0.590789 seconds to LoadMap(/Game/DeepDrive/Maps/DeepDriveSim_Demo)
[2020.12.31-18.22.22:962][  0]LogSlate: Took 0.000036 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoBold.ufont' (160K)
libgcc_s.so.1 must be installed for pthread_cancel to work
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=68112 LargeMemoryPoolOffset=199248 
Malloc Size=307856 LargeMemoryPoolOffset=507120 
Malloc Size=72403 LargeMemoryPoolOffset=579539 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)
```

Heavy02011, 2020-12-31:
Any chance to run it in docker?