deepdrive/deepdrive #51: Installation issue

hhassannejad, 2020-01-07:
Hello,

I am trying to install deepdrive on ubuntu 16.04, using Conda, python 3.6, and Tensorflow 1.9.

But, the installation fails after demonstration following messages:

> Checking python version...check!
Checking for valid Tensorflow installation
/home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:523: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:524: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:526: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:527: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:532: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
Tensorflow 1.9.0 detected - meets min version (1.7)
running command: /home/hhassannejad/anaconda3/envs/deepdrive/bin/python -m pip install sarge wheel requests boto
Requirement already satisfied: sarge in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (0.1.5.post0)
Requirement already satisfied: wheel in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (0.33.6)
Requirement already satisfied: requests in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (2.22.0)
Requirement already satisfied: boto in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (2.49.0)
Requirement already satisfied: idna<2.9,>=2.5 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from requests) (2.8)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from requests) (3.0.4)
Requirement already satisfied: certifi>=2017.4.17 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from requests) (2019.11.28)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from requests) (1.25.7)
Requirement already satisfied: scipy==1.2.1 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (1.2.1)
Requirement already satisfied: numpy>=1.16.1 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 2)) (1.17.4)
Requirement already satisfied: h5py in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 3)) (2.10.0)
Requirement already satisfied: gym==0.10.0 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 4)) (0.10.0)
Requirement already satisfied: arrow in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 5)) (0.15.5)
Requirement already satisfied: matplotlib==3.0.3 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 6)) (3.0.3)
Requirement already satisfied: pytest-xdist in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 7)) (1.31.0)
Requirement already satisfied: requests in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 8)) (2.22.0)
Requirement already satisfied: clint in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 9)) (0.5.1)
Requirement already satisfied: Pillow==6.2.0 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 10)) (6.2.0)
Requirement already satisfied: boto in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 11)) (2.49.0)
Requirement already satisfied: future==0.16.0 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 12)) (0.16.0)
Requirement already satisfied: pyzmq in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 13)) (18.1.1)
Requirement already satisfied: gputil in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 14)) (1.4.0)
Requirement already satisfied: joblib in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 15)) (0.14.1)
Requirement already satisfied: cloudpickle in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 16)) (1.2.2)
Requirement already satisfied: pyarrow==0.12.1 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 17)) (0.12.1)
Requirement already satisfied: pyglet in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 18)) (1.4.9)
Requirement already satisfied: six in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 19)) (1.13.0)
Requirement already satisfied: flask in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 20)) (1.1.1)
Requirement already satisfied: cherrypy in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 21)) (18.5.0)
Requirement already satisfied: paste in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 22)) (3.2.3)
Requirement already satisfied: simple-pid in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 23)) (0.2.4)
Requirement already satisfied: python-box in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 25)) (4.0.4)
Requirement already satisfied: sarge in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 26)) (0.1.5.post0)
Requirement already satisfied: deepdrive-api>=3.1.20191111204349 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 27)) (3.1.20191111204349)
Requirement already satisfied: ue4_ci_helpers>=0.0.9 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 28)) (0.0.11)
Requirement already satisfied: ue4cli in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 29)) (0.0.39)
Requirement already satisfied: python-gist==0.6.1.dev42 from git+git://github.com/deepdrive/gist.git#egg=python-gist==0.6.1.dev42 in /home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/site-packages (from -r requirements.txt (line 36)) (0.6.1.dev42)
[... cut ...]
>   File "/home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/ssl.py", line 817, in __init__
>     self.do_handshake()
>   File "/home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/ssl.py", line 1077, in do_handshake
>     self._sslobj.do_handshake()
>   File "/home/hhassannejad/anaconda3/envs/deepdrive/lib/python3.6/ssl.py", line 689, in do_handshake
>     self._sslobj.do_handshake()
> ssl.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)
> 

Is there a certain certificate to install? or a trusted-host to introduce? thank you!

crizCraig, 2020-01-10:
Hmm, this is working from my side. Perhaps a retry would work... This is connecting to an AWS S3 bucket `deepdrive`. Also see [here](https://stackoverflow.com/questions/28115250/boto-ssl-certificate-verify-failed-certificate-verify-failed-while-connecting) for some possible workarounds.

hhassannejad, 2020-01-15:
Thank you Craig! 
Adding _is_secure=False_ to the connection solved this particular problem, however I needed to fix a few other similar problems to be able to run the examples. Although, I need to address the problem in a more stable way, it is fine for now. Thank you!

crizCraig, 2020-01-15:
Great to hear! Sorry about needing to set _is_secure=False_ though. Not sure why your installed certs would not contain the one S3 uses. Hopefully this is resolved soon!

closed 2020-01-15 by crizCraig