deepdrive/deepdrive #54: RuntimeError: Could not find a sim version matching sim/deepdrive-sim-windows-3.1

shott92, 2020-02-06:
When trying to run `python example.py` I keep getting the following runtime error;
```
--------- Updating to latest simulator ----------
2020-02-06 23:31:40,761 - util.ensure_sim - INFO - Downloading latest sim
Traceback (most recent call last):
  File "example_sync.py", line 38, in <module>
    main()
  File "example_sync.py", line 25, in main
    randomize_month=False)
  File "C:\Users\x\deepdrive\sim\__init__.py", line 49, in start
    env = start_local_env(args)
  File "C:\Users\x\deepdrive\sim\__init__.py", line 62, in start_local_env
    env = gym.make(args.env_id)
  File "C:\Users\x\Miniconda3\envs\dd\lib\site-packages\gym\envs\registration.py", line 163, in make
    return registry.make(id)
  File "C:\Users\x\Miniconda3\envs\dd\lib\site-packages\gym\envs\registration.py", line 119, in make
    env = spec.make()
  File "C:\Users\x\Miniconda3\envs\dd\lib\site-packages\gym\envs\registration.py", line 86, in make
    env = cls(**self._kwargs)
  File "C:\Users\x\deepdrive\sim\gym_env.py", line 125, in __init__
    util.ensure_sim.ensure_sim()
  File "C:\Users\x\deepdrive\util\ensure_sim.py", line 32, in ensure_sim
    url = c.AWS_BUCKET_URL + get_latest_sim_url()
  File "C:\Users\x\deepdrive\util\ensure_sim.py", line 139, in get_latest_sim_url
    'in bucket %s' % (bucket_search_str, c.AWS_BUCKET_URL))
RuntimeError: Could not find a sim version matching sim/deepdrive-sim-windows-3.1 in bucket https://s3-us-west-1.amazonaws.com/deepdrive
```
System config;
Miniconda with python3.7
TensorFlow 1.15.2 / Tensorflow_hub 0.7.0
Windows 10
No GPU (well, Radeon 7470M)

Thanks

Teo03, 2020-04-26:
I also get the same results, it seems that this is a windows-only error. Please help!

crizCraig, 2020-04-27:
Hi @Teo03 - Unfortunately, we are not distributing Windows builds due to the lack of cloud build infra (Linux is built and uploaded automatically on GCP). We do use Windows for development though, and so you could setup the project via the development build instructions which would give you access to Unreal Editor as well.

https://simdocs.deepdrive.io/docs/setup/windows

Teo03, 2020-04-27:
@crizCraig thanks!