deepdrive/deepdrive #59: Control over PID Path Follower's Speed

raks097, 2020-03-05:
Is there a way one can control/limit the speed at which the path-follower agent travels at?
I have gone through the source code and it seems like the actuation of the vehicles is done through deepdrive_client.advance_synchronous_stepping(), but since it comes a pre-compiled package, I haven't found ways in which we can manipulate the PID bot's actuations.

crizCraig, 2020-03-05:
Hi @raks097 - the pid bot is in C++ baked into the Unreal binary, so it's more difficult to modify. It's meant to be more for NPC's than for bots that people modify. You can however easily play with the speed and several other config variables in the Unreal Editor if you setup the [sim](https://github.com/deepdrive/deepdrive-sim) for [development](https://simdocs.deepdrive.io/docs/setup/linux). The underlying C++ for that agent is [here](/tree/master/Plugins/DeepDrivePlugin/Source/DeepDrivePlugin/Private/Simulation/Traffic/BehaviorTree) which you can also modify.

If you want to submit such an agent to the leaderboard, we'll need to go through a special process as you are modifying the sim project itself. I.e. we'd need to review your code changes and merge them into the main project - at which point you would be improving the _deepdrive_ agent as well. Although we would provide attribution to you if you improve performance!

raks097, 2020-03-06:
@crizCraig 
Thanks for the quick reply. 
Will look into the setting up the Unreal Editor for further modifications.
However, the link that you provided for the C++ bot seems to be broken.
Could you please give a working link ?

Thanks.

crizCraig, 2020-03-06:
Fixed above, and here again https://github.com/deepdrive/deepdrive-sim/tree/master/Plugins/DeepDrivePlugin/Source/DeepDrivePlugin/Private/Simulation/Traffic/BehaviorTree
Thanks @raks097 !