deepdrive/deepdrive #61: Error running install.py

DylanMashini, 2020-05-22:
* Install failed, fix issues below and rerun install.py 

Traceback (most recent call last):
  File "install.py", line 237, in <module>
    raise e
  File "install.py", line 233, in <module>
    main()
  File "install.py", line 123, in main
    import config as c
  File "/Users/dylanmashini/deepdrive/config/__init__.py", line 1, in <module>
    from config.directories import *
  File "/Users/dylanmashini/deepdrive/config/directories.py", line 1, in <module>
    from util.get_directories import *
  File "/Users/dylanmashini/deepdrive/util/get_directories.py", line 71, in <module>
    DEEPDRIVE_DIR = _get_deepdrive_dir()
  File "/Users/dylanmashini/deepdrive/util/get_directories.py", line 52, in _get_deepdrive_dir
    os.makedirs(ret)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/os.py", line 223, in makedirs
    mkdir(name, mode)
FileNotFoundError: [Errno 2] No such file or directory: ''

I get this error running install.py on MacOS 10.15.1 with python 3.7

crizCraig, 2020-05-22:
Unfortunately, this project only supports Linux. Deepdrive-zero should be
installable on Mac though as it’s main dependencies are arcade, gym,
pytorch, and numba. Also much faster to iterate on control problems as it
runs hundreds of times faster than real-time.

DylanMashini, 2020-05-22:
OK, thanks. I'm going to close this issue now.

closed 2020-05-22 by DylanMashini