deepdrive/deepdrive #62: Deepdrive client is not able to connect to the environment

fidelechevarria, 2020-06-24:
Hi DeepDrive team! Congratulations on this awesome project. I would appreciate some help regarding an issue I'm facing in my setup.

## Description
My goal is to run the file `example.py`. It seems that the DeepDrive client is not able to connect to the environment, so the car doesn't move in the simulator. The simulator runs fine and I am able to control the car manually, but the constant throttle commanded in file `example.py` has no effect.

## My specs
```
Ubuntu 18.04.4 LTS
Python 3.6.10
Unreal Engine 4.21
DeepDrive 3.1, master branch
```
## Steps to reproduce
* Install DeepDrive
* Run `python example.py`

## Log messages
First 30 lines:
```
(deepdrive) fidel@fidel-desktop:~/repos/deepdrive$ python example.py 
### ><>|><> PyInit_deepdrive_client <><|<>< ###
#### ><> ><> PyInit_deepdrive_simulation <>< <>< ####
/home/fidel/anaconda3/envs/deepdrive/lib/python3.6/site-packages/gym/envs/registration.py:14: PkgResourcesDeprecationWarning: Parameters to load are deprecated.  Call .resolve and .require separately.
  result = entry_point.load(False)
## ><>|><> PyInit_deepdrive_capture <><|<>< ##
Init numPy for PyCaptureCameraObject
Init numPy for PyCaptureLastCollisionObject
Init numPy for PyCaptureSnapshotObject
Warning, using parent logger to avoid nested loggers and duplicate errors messages
Warning, using parent logger to avoid nested loggers and duplicate errors messages
Warning, using parent logger to avoid nested loggers and duplicate errors messages
Address successfully parsed 127.0.0.1:9876
Couldn't connect to 127.0.0.1:9876
Requirement already satisfied: pyzmq in /home/fidel/Deepdrive/deepdrive-sim-linux-3.1.20200229010300/Engine/Plugins/UnrealEnginePython/EmbeddedPython/Linux/lib/python3.6/site-packages (18.0.1)
Requirement already satisfied: pyarrow==0.12.1 in /home/fidel/Deepdrive/deepdrive-sim-linux-3.1.20200229010300/Engine/Plugins/UnrealEnginePython/EmbeddedPython/Linux/lib/python3.6/site-packages (0.12.1)
Requirement already satisfied: requests in /home/fidel/Deepdrive/deepdrive-sim-linux-3.1.20200229010300/Engine/Plugins/UnrealEnginePython/EmbeddedPython/Linux/lib/python3.6/site-packages (2.21.0)
Requirement already satisfied: six>=1.0.0 in /home/fidel/Deepdrive/deepdrive-sim-linux-3.1.20200229010300/Engine/Plugins/UnrealEnginePython/EmbeddedPython/Linux/lib/python3.6/site-packages (from pyarrow==0.12.1) (1.15.0)
Requirement already satisfied: numpy>=1.14 in /home/fidel/Deepdrive/deepdrive-sim-linux-3.1.20200229010300/Engine/Plugins/UnrealEnginePython/EmbeddedPython/Linux/lib/python3.6/site-packages (from pyarrow==0.12.1) (1.16.3)
Requirement already satisfied: idna<2.9,>=2.5 in /home/fidel/Deepdrive/deepdrive-sim-linux-3.1.20200229010300/Engine/Plugins/UnrealEnginePython/EmbeddedPython/Linux/lib/python3.6/site-packages (from requests) (2.8)
Requirement already satisfied: urllib3<1.25,>=1.21.1 in /home/fidel/Deepdrive/deepdrive-sim-linux-3.1.20200229010300/Engine/Plugins/UnrealEnginePython/EmbeddedPython/Linux/lib/python3.6/site-packages (from requests) (1.24.2)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /home/fidel/Deepdrive/deepdrive-sim-linux-3.1.20200229010300/Engine/Plugins/UnrealEnginePython/EmbeddedPython/Linux/lib/python3.6/site-packages (from requests) (3.0.4)
Requirement already satisfied: certifi>=2017.4.17 in /home/fidel/Deepdrive/deepdrive-sim-linux-3.1.20200229010300/Engine/Plugins/UnrealEnginePython/EmbeddedPython/Linux/lib/python3.6/site-packages (from requests) (2019.3.9)
You are using pip version 18.1, however version 20.2b1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
2020-06-24 23:36:17,499 - util.ensure_sim - INFO - Installed UEPy python dependencies
2020-06-24 23:36:17,511 - sim - INFO - No open sim detected
2020-06-24 23:36:18,527 - sim - INFO - Starting simulator with ['/home/fidel/Deepdrive/deepdrive-sim-linux-3.1.20200229010300/DeepDrive/Binaries/Linux/DeepDrive', '/DeepDriveKevindalePlugin/DeepDrive/Maps/DeepDriveSim_Kevindale_Bare', '-remote_ai', '-scenario_mode', '-scenario_index=1', '-opengl4', '-ResX=1280', '-ResY=720', '-WINDOWED', '-WinX=-1280'] (takes a few seconds the first time).
2020-06-24 23:36:18,528 - sim - WARNING - Connection to environment failed, retry (2/10) in 2 seconds
Increasing per-process limit of core file size to infinity.
```
It seems that call to `deepdrive_client.create('127.0.0.1', 9876)` times out and is retried 10 times until eventually raising runtime error `Could not connect to the environment`. I would appreciate some advice on how to fix this issue. Please let me know if you need anything else from my side. Many thanks!

crizCraig, 2020-06-29:
I think I am able to reproduce @fidelechevarria , but could you send the entire log (perhaps in a gist) just to be sure?

crizCraig, 2020-06-29:
@fidelechevarria Try pulling latest where we lock to a known working version of numpy. Problem on my side was with the latest `1.19.0` numpy causing segfaults in the shared memory read.

crizCraig, 2020-06-30:
To clarify, if you want to modify your existing installation and python environment, you can just

```
pip uninstall numpy
git pull
python install.py  # or just `pip install numpy==1.16.1`
```