deepdrive/deepdrive #66: pyarrow breaks the install.py using python3.9

shyamalschandra, 2020-11-08:
What should I do?  Is there some timeline for a diff and patch?

shyamalschandra, 2020-11-08:
```
root@77d2c138cf98:/home/shyamal/deepdrive# python3.9 -m pip install -r requirements.txt 
Collecting scipy
  Using cached scipy-1.5.4-cp39-cp39-manylinux1_x86_64.whl (25.8 MB)
Requirement already satisfied: numpy in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 2)) (1.19.4)
Collecting h5py
  Using cached h5py-3.1.0-cp39-cp39-manylinux1_x86_64.whl (4.4 MB)
Collecting gym
  Using cached gym-0.17.3.tar.gz (1.6 MB)
Collecting arrow
  Using cached arrow-0.17.0-py2.py3-none-any.whl (50 kB)
Requirement already satisfied: pytest-xdist in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 6)) (2.1.0)
Collecting requests
  Using cached requests-2.24.0-py2.py3-none-any.whl (61 kB)
Collecting clint
  Using cached clint-0.5.1.tar.gz (29 kB)
Requirement already satisfied: Pillow in /usr/lib/python3/dist-packages (from -r requirements.txt (line 9)) (3.1.2)
Collecting boto
  Using cached boto-2.49.0-py2.py3-none-any.whl (1.4 MB)
Collecting future
  Using cached future-0.18.2.tar.gz (829 kB)
Collecting pyzmq
  Using cached pyzmq-19.0.2-cp39-cp39-manylinux1_x86_64.whl (1.1 MB)
Collecting gputil
  Using cached GPUtil-1.4.0.tar.gz (5.5 kB)
Collecting joblib
  Using cached joblib-0.17.0-py3-none-any.whl (301 kB)
Collecting cloudpickle
  Using cached cloudpickle-1.6.0-py3-none-any.whl (23 kB)
Collecting pyarrow
  Using cached pyarrow-1.0.1.tar.gz (1.3 MB)
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3.9 /usr/local/lib/python3.9/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-888juo_9/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'cython >= 0.29' 'numpy==1.14.5; python_version<'"'"'3.7'"'"'' 'numpy==1.16.0; python_version>='"'"'3.7'"'"'' setuptools setuptools_scm wheel
       cwd: None
  Complete output (710 lines):
  Ignoring numpy: markers 'python_version < "3.7"' don't match your environment
  Collecting cython>=0.29
    Using cached Cython-0.29.21-cp39-cp39-manylinux1_x86_64.whl (1.9 MB)
  Collecting numpy==1.16.0
    Using cached numpy-1.16.0.zip (5.1 MB)
  Collecting setuptools
    Using cached setuptools-50.3.2-py3-none-any.whl (785 kB)
  Collecting setuptools_scm
    Using cached setuptools_scm-4.1.2-py2.py3-none-any.whl (27 kB)
  Collecting wheel
    Using cached wheel-0.35.1-py2.py3-none-any.whl (33 kB)
  Building wheels for collected packages: numpy
    Building wheel for numpy (setup.py): started
    Building wheel for numpy (setup.py): finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-ma35s7ou/numpy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-ma35s7ou/numpy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-sl6959pr
         cwd: /tmp/pip-install-ma35s7ou/numpy/
    Complete output (336 lines):
    Running from numpy source directory.
    /tmp/pip-install-ma35s7ou/numpy/numpy/distutils/misc_util.py:476: SyntaxWarning: "is" with a literal. Did you mean "=="?
      return is_string(s) and ('*' in s or '?' is s)
    /usr/lib/python3.9/distutils/dist.py:274: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)
    blas_opt_info:
    blas_mkl_info:
    customize UnixCCompiler
      libraries mkl_rt not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/x86_64-linux-gnu']
      NOT AVAILABLE
  
    blis_info:
    customize UnixCCompiler
      libraries blis not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/x86_64-linux-gnu']
      NOT AVAILABLE
  
[... 654 lines cut ...]
          self.run_command(cmd_name)
        File "/usr/lib/python3.9/distutils/cmd.py", line 313, in run_command
          self.distribution.run_command(command)
        File "/usr/lib/python3.9/distutils/dist.py", line 985, in run_command
          cmd_obj.run()
        File "/tmp/pip-install-ma35s7ou/numpy/numpy/distutils/command/build_src.py", line 148, in run
          self.build_sources()
        File "/tmp/pip-install-ma35s7ou/numpy/numpy/distutils/command/build_src.py", line 165, in build_sources
          self.build_extension_sources(ext)
        File "/tmp/pip-install-ma35s7ou/numpy/numpy/distutils/command/build_src.py", line 322, in build_extension_sources
          sources = self.generate_sources(sources, ext)
        File "/tmp/pip-install-ma35s7ou/numpy/numpy/distutils/command/build_src.py", line 375, in generate_sources
          source = func(extension, build_dir)
        File "numpy/core/setup.py", line 423, in generate_config_h
          moredefs, ignored = cocache.check_types(config_cmd, ext, build_dir)
        File "numpy/core/setup.py", line 47, in check_types
          out = check_types(*a, **kw)
        File "numpy/core/setup.py", line 279, in check_types
          raise SystemError(
      SystemError: Cannot compile 'Python.h'. Perhaps you need to install python-dev|python-devel.
      ----------------------------------------
  ERROR: Command errored out with exit status 1: /usr/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-ma35s7ou/numpy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-ma35s7ou/numpy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"' [... cut ...]
  ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3.9 /usr/local/lib/python3.9/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-888juo_9/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'cython >= 0.29' 'numpy==1.14.5; python_version<'"'"'3.7'"'"'' 'numpy==1.16.0; python_version>='"'"'3.7'"'"'' setuptools setuptools_scm wheel Check the logs for full command output.
```

Rookie-81, 2020-11-30:
i have the same issue

rahulbhadani, 2021-03-14:
I have the same issue with python 3.8

mfaccin, 2022-01-25:
same here. does anyone have a fix for it?