deepdrive/deepdrive #73: Installation failed with AWS S3 returning 403

WeberSamuel, 2021-09-10:
Hi,
I tried to install deepdrive on Windows but after all packages have been setup the installation fails with this error message:

Install failed, fix issues below and rerun install.py
Traceback (most recent call last):
  File "install.py", line 237, in <module>
    raise e
  File "install.py", line 233, in <module>
    main()
  File "install.py", line 134, in main
    bindings_version = get_latest_valid_bindings()
  File "install.py", line 73, in get_latest_valid_bindings
    bucket = conn.get_bucket('deepdrive')
  File "C:\Users\samue\anaconda3\envs\deepdrive\lib\site-packages\boto\s3\connection.py", line 509, in get_bucket
    return self.head_bucket(bucket_name, headers=headers)
  File "C:\Users\samue\anaconda3\envs\deepdrive\lib\site-packages\boto\s3\connection.py", line 542, in head_bucket
    raise err
boto.exception.S3ResponseError: S3ResponseError: 403 Forbidden

I used anaconda with python 3.7.

dks2000dks, 2021-11-13:
Yes, I am also facing the same issue. Does this have a solution?

KailoPictza, 2023-02-09:
I am new to all of this, but with some searching around for a solution I found this link [(https://s3-us-west-1.amazonaws.com/deepdrive)]:

`This XML file does not appear to have any style information associated with it. The document tree is shown below.
<Error>
<link type="text/css" rel="stylesheet" id="dark-mode-custom-link"/>
<link type="text/css" rel="stylesheet" id="dark-mode-general-link"/>
<style lang="en" type="text/css" id="dark-mode-custom-style"/>
<style lang="en" type="text/css" id="dark-mode-native-style"/>
<Code>NoSuchBucket</Code>
<Message>The specified bucket does not exist</Message>
<BucketName>deepdrive</BucketName>
<RequestId>Z4Y9D4FN38EP1NEH</RequestId>
<HostId>JK5trJwHrO5JARJi0K10RbB53+VK7cCmBEMr58h9jjYdx4bwvVR6QQNjkhbAGrBIOEOMOOjkmZ4=</HostId>
</Error>`

xieshangfeiabc, 2024-07-23:
I am also facing the same issue. Does this have a solution?